devbisme/KiPart #1: Error when loading part generated by kipart

reportingsjr, 2015-08-17:
Hi,

I tried to generate a single part with kipart and when I add it to the libraries list I get the following error:
"Library 'lt1512' component load error error <error 'pin only had 6 parameters of the required 11 or 12' in DRAW command X> occurred at line 15 ."

Here is the csv I created: https://dl.dropboxusercontent.com/u/9135021/lt1512.csv
Here is the lib that was created: https://dl.dropboxusercontent.com/u/9135021/lt1512.lib

The command I used was `kipart lt1512.csv -o lt1512.lib`

Running this on fedora 22 with the latest revision of kicad (bzr 6089).

xesscorp, 2015-08-17:
Good error report. Thanks! Fixed.

closed 2015-08-17 by xesscorp

reportingsjr, 2015-08-17:
Works great now. Thank you! Cool tool, hopefully something like this can be further integrated in to KiCad in the future.

commit 0b7895ed (2015-08-17): Fixed #1 - Bumped version number.