devbisme/KiPart #4: Changing the symbol origin

xesscorp, 2015-09-16:
Allow KiPart to place the symbol origin at pin #1 or the centroid of the symbol.

inkblotter, 2016-09-19:
+1 I think it should be at the centroid of the symbol

hyOzd, 2016-09-19:
> +1 I think it should be at the centroid of the symbol

I believe this is also the kicad library convention. It's not mentioned in the [KLC](https://github.com/KiCad/kicad-library/wiki/Kicad-Library-Convention) directly but I was warned by librarians about this.

GeoSpark, 2016-12-10:
It is mentioned in the KLC [here](https://github.com/KiCad/kicad-library/wiki/Kicad-Library-Convention#3-general-rules-for-symbols), item 3 "Origin is placed in the middle of symbol."

I have written a fix for this issue, but the problem is with the placement of the Ref and Part fields for multi-unit parts. Currently this is fine because the origin is always at pin 1, so the labels are always in the same place. My fix just translates the centre of each unit to the origin, but if two units have different heights then the labels will either be far away or overlapping the symbol.

My idea is to use the centroid of the first unit as a fixed offset for all other units. So taller units will hang down below the x axis, and shorter units will float above it.

Thoughts?

xesscorp, 2016-12-13:
I'm not sure if it's worth the effort to meet this requirement. Having the origin at the centroid doesn't fix any real-world problems and we can't even meet the requirement if we have multi-unit symbols (as you stated).

Go ahead and submit a pull request for your fix if you want. I'll look at it. If it makes the drawing code even more opaque, then I probably won't accept it. If it is accepted, there needs to be a new kipart option called '--centroid' that enables the centroid as origin and the option has to be off by default.

GeoSpark, 2016-12-14:
The fix is fairly simple, so shouldn't obfuscate the code too much, but I totally understand if you don't want to merge it - I won't be offended :)

It'll probably be after the weekend before I have chance to tidy it up and submit it.