devbisme/KiPart #7: * at start of pin number not generating "gap" pin

jhelwig, 2016-02-16:
The [usage documentation](https://kipart.readthedocs.org/en/latest/usage.html) states that putting a `*` at the start of a pin number will create a non-exxistent "gap" pin for visual grouping, when using `-s row`, but this doesn't appear to be the case.

When I run `kipart -r generic -s row -o printer_parts.lib --overwrite kipart_parts.csv` with KiPart 0.1.14 using the following CSV:
```
X95840WV20I-2.7
Pin,Type,Style,Side,Name
14,input,,left,A0
15,input,,left,A1
4,input,,left,A2
*5,io,clock,left,SCL
6,io,,left,SDA
17,input,,left,~WP
*16,pwr,,left,VCC
7,pwr,,left,GND
2,passive,,right,RL3
3,output,,right,RW3
1,passive,,right,RH3
*9,passive,,right,RL2
8,output,,right,RW2
10,passive,,right,RH2
*12,passive,,right,RL1
11,output,,right,RW1
13,passive,,right,RH1
*19,passive,,right,RL0
20,output,,right,RW0
18,passive,,right,RH0
```

I get the following part:
[image]

It would be nice to have an example of generating the "gap" pins in https://kipart.readthedocs.org/en/latest/usage.html in case this isn't the proper format that should be generating "gap" pins.

xesscorp, 2016-02-18:
Thanks for the clear bug report. I think I have this fixed. Try the newest version 0.1.15 and let me know.

jhelwig, 2016-02-18:
:+1: 0.1.15 Is generating the gap pins as expected. Thanks!

[image]

closed 2016-03-20 by xesscorp