devbisme/KiPart #11: multiple parts in a single csv file doesn't work

hyOzd, 2016-06-14:
From the instructions in usage manual I've created this CSV file;

```
CONN_03X02

Pin, Unit
1, 1
2, 1
3, 1
4, 2
5, 2
6, 2

CONN_03X03

Pin, Unit
1, 1
2, 1
3, 1
4, 2
5, 2
6, 2
7, 3
8, 3
9, 3
```

But I got this error:

```
Traceback (most recent call last):
  File "/usr/local/bin/kipart", line 9, in <module>
    load_entry_point('kipart==0.1.16', 'console_scripts', 'kipart')()
  File "/usr/local/lib/python2.7/dist-packages/kipart-0.1.16-py2.7.egg/kipart/__main__.py", line 127, in main
    append_to_lib = call_kipart(part_data_file)
  File "/usr/local/lib/python2.7/dist-packages/kipart-0.1.16-py2.7.egg/kipart/__main__.py", line 112, in call_kipart
    debug_level=args.debug)
  File "/usr/local/lib/python2.7/dist-packages/kipart-0.1.16-py2.7.egg/kipart/kipart.py", line 544, in kipart
    for part_num, pin_data in part_reader(part_data_file):
  File "/usr/local/lib/python2.7/dist-packages/kipart-0.1.16-py2.7.egg/kipart/generic_reader.py", line 88, in generic_reader
    setattr(pin, a, fix_pin_data(row[c], part_num))
  File "/usr/local/lib/python2.7/dist-packages/kipart-0.1.16-py2.7.egg/kipart/common.py", line 118, in fix_pin_data
    fixed_pin_data = pin_data.strip()  # Remove leading/trailing spaces.
AttributeError: 'NoneType' object has no attribute 'strip'
```

After some investigations I've noticed that blank line separating the parts is ignored because `csv.DictReader` used [here](https://github.com/xesscorp/KiPart/blob/6d2aee6519938979a19ae6a44d9f2a3da3bfe286/kipart/generic_reader.py#L74) simply skips empty lines. [This](http://stackoverflow.com/questions/27707581/why-does-csv-dictreader-skip-empty-lines) stackoverflow question has some info on the matter. 

As a workaround I've used a line containing a single comma to separate parts. Ex:

```
...
5, 2
6, 2
,
CONN_03X03

Pin, Unit
1, 1
2, 1
...
```

xesscorp, 2016-06-18:
I verfified this problem. I never noticed it before because I create CSV files using LibreOffice Calc which  places a row of commas for every blank line in the spreadsheet.

xesscorp, 2018-01-03:
This issue was fixed on 05/24/2017 but it didn't close at that time.

closed 2018-01-03 by xesscorp