devbisme/KiPart #18: Error creating symbol

PTaylor-RA, 2016-10-14:
I created the attached CSV and when I run the command line below I get this console output:

`kipart --sort row --bundle --overwrite --output mcu.lib ATMEL_ATSAMA5D27.csv`

`Traceback (most recent call last):
  File "/usr/local/bin/kipart", line 9, in <module>
    load_entry_point('kipart==0.1.21', 'console_scripts', 'kipart')()
  File "/usr/local/lib/python2.7/dist-packages/kipart-0.1.21-py2.7.egg/kipart/__main__.py", line 144, in main
    append_to_lib = call_kipart(part_data_file)
  File "/usr/local/lib/python2.7/dist-packages/kipart-0.1.21-py2.7.egg/kipart/__main__.py", line 127, in call_kipart
    debug_level=args.debug)
  File "/usr/local/lib/python2.7/dist-packages/kipart-0.1.21-py2.7.egg/kipart/kipart.py", line 571, in kipart
    fuzzy_match=fuzzy_match)
  File "/usr/local/lib/python2.7/dist-packages/kipart-0.1.21-py2.7.egg/kipart/kipart.py", line 487, in draw_symbol
    draw_pins(lib_file, unit_num, sorted_side_pins, transform[side],
KeyError: ''`

and the attached .lib
[symbol.zip](https://github.com/xesscorp/KiPart/files/529942/symbol.zip)

PTaylor-RA, 2016-10-15:
Same error with this CSV
[Micron_MTFC2GMDEA-0M_WT.csv.zip](https://github.com/xesscorp/KiPart/files/531490/Micron_MTFC2GMDEA-0M_WT.csv.zip)

GeoSpark, 2016-12-10:
Your first issue is because the "side" field is missing for the pin G12 DDR_CAS. Granted it shouldn't crash, but the fix should be straightforward.

Your second issue is essentially the same, but you do have just an asterisk for the pin number on some rows, with no name or anything.

closed 2017-05-22 by xesscorp