devbisme/KiPart #24: error: pin has 10 parameters of the required 11 or 12 in DRAW command X

nm111, 2017-07-28:
I have a CSV that I converted to a .lib with `kipart lm80-p0598-1_c_apq8064e_datasheet2.csv -s num -o snapdragon.lib`, which gives no errors or warnings, but when imporing to KiCAD, I get the following error:
`Library 'file' component load error error <error: pin has 10 parameters of the required 11 or 12 in DRAW command X> occured at line 1095`

I am running KiPart 0.1.28 on MacOS Sierra 10.12.3.

[file.lib.zip](https://github.com/xesscorp/KiPart/files/1183638/file.lib.zip)

[lm80-p0598-1_c_apq8064e_datasheet2.csv.zip](https://github.com/xesscorp/KiPart/files/1183631/lm80-p0598-1_c_apq8064e_datasheet2.csv.zip)

xesscorp, 2017-07-28:
The immediate solution to your problem is to place a part identifier for your part in the first row of your CSV file. That should be followed by the row with "Pin,Name". And then the remaining rows with your part pin data. Then the symbol will be created correctly.

nm111, 2017-07-28:
Oh. Sorry, I forgot that.

closed 2017-07-28 by nm111