devbisme/KiPart #25: default output directory should be working directory

hyOzd, 2017-07-31:
I've installed kipart to system directories using pip. When I run kipart without an output path, I get this error.

```
Traceback (most recent call last):
  File "/usr/local/bin/kipart", line 9, in <module>
    load_entry_point('kipart==0.1.28', 'console_scripts', 'kipart')()
  File "/usr/local/lib/python2.7/dist-packages/kipart/__main__.py", line 144, in main
    append_to_lib = call_kipart(part_data_file)
  File "/usr/local/lib/python2.7/dist-packages/kipart/__main__.py", line 127, in call_kipart
    debug_level=args.debug)
  File "/usr/local/lib/python2.7/dist-packages/kipart/kipart.py", line 660, in kipart
    with open(lib_filename, lib_filemode) as lib_file:
IOError: [Errno 13] Permission denied: '/usr/local/bin/kipart.lib'
```

I think it happens because of [this line of code](https://github.com/xesscorp/KiPart/blob/master/kipart/__main__.py#L107). Path of kipart executable, is used for default directory. I would love to change this to the current working directory if you agree.

xesscorp, 2017-07-31:
You're right that kipart should create a default file in the current directory rather than the executable directory. I was already getting a new release ready to correct another error, so I added that change as well. The default library `kipart.lib` will be created in the current directory if no output library is specified. (I also changed `kilib2csv.py` to use `kipart.csv` as its default output file for the same reason.)

Feel free to change this if you can think of a better method.

hyOzd, 2017-07-31:
That's what I had in mind as well. Thank you.

closed 2017-07-31 by hyOzd