devbisme/KiPart #26: JSON?

beriberikix, 2017-08-04:
Have you also considered supporting JSON in addition to CSV?

xesscorp, 2017-08-05:
You would need some predefined schema for the structure of the JSON file. Then it could be translated into a CSV file and passed to KiPart, possibly using a tool like [this](https://github.com/evidens/json2csv). Then it would all have to be documented.

What's the use case for this feature? Generating and manipulating pin data seems much more cumbersome in JSON with a text editor versus CSV using a spreadsheet. Is there some tool or database for parts that dumps out JSON?

beriberikix, 2017-08-06:
My use case is an idea I've been thinking about for awhile - a database for pinouts (loosely-described [here](https://hackaday.io/project/17984-whichpincom).) Manipulating JSON on a server is far more efficient and will make it more database-friendly since I'm using a JSON-based DB.

I'd imagine this would require a significant rework of this project but it moved to an intermediate representation format it would allow KiPart to support more input formats (and even output ones,)

xesscorp, 2017-08-06:
Your database looks like it will contain more than just pin information. It's probably easiest for you to convert your JSON into the restricted CSV format used by KiPart. Otherwise, KiPart will need to have an understanding of your database schema in order to pull out the information it needs. That schema will probably change as you develop things. The KiPart CSV format hasn't changed in years so it's a stable target to aim at. And the computational effort to do the conversion is the same whether it happens on the database side or inside KiPart.

closed 2018-01-03 by xesscorp