devbisme/KiPart #31: Pin Ordering Question

oafilipoai, 2017-12-05:
Hello,

I'm using the following test.csv file to create a symbol:

```
TestIC,,,,,,
,,,,,,
Pin,Unit,Name,Type,Side,Style,Hidden
A1,,FUNC1_IN,in,left,,
A2,,FUNC1_IN,in,left,,
B1,,FUNC1_OUT,out,right,,
B2,,FUNC1_OUT,out,right,,
C1,,FUNC2_IN,in,left,,
C2,,FUNC2_IN,in,left,,
D1,,FUNC2_OUT,out,right,,
D2,,FUNC2_OUT,out,right,,
E1,,FUNC3_IN,in,left,,
E2,,FUNC3_IN,in,left,,
F1,,FUNC3_OUT,out,right,,
F2,,FUNC3_OUT,out,right,,
```

Then I run kipart with default options:

    kipart test.csv

The pin order on the left side is as expected (FUNC1, FUNC2, FUNC3) but on the right side it is reversed (FUNC3, FUNC2, FUNC1 from top to bottom)

How can I keep the pins on the left and right sides aligned (e.g. FUNC1 pins above FUNC2 pins on both sides)? Am I missing something in the documentation?

xesscorp, 2017-12-07:
Currently, kipart proceeds around the chip in a counter-clockwise direction. So when it gets to the right side, it starts placing pins at the bottom and proceeds upward. You'll have to reverse the order of the right-side pins or else rearrange the pins in the symbol editor.

oafilipoai, 2017-12-07:
Thanks for the reply. I ended up mirroring the right side pins in the symbol editor.

closed 2017-12-07 by oafilipoai