devbisme/KiPart #34: _csv.Error: new-line character seen in unquoted field

bullitbd, 2018-02-19:
getting
`_csv.Error: new-line character seen in unquoted field - do you need to open the file in universal-newline mode?`
osx 10.11, Office (Excel) for Mac 2011., Python 2.7; attempted saving as Mac, Windows, MS-DOS format as many variously have suggested. Hand-built the file in text editor - there are no newline characters, but get same error.

Here is file:
`Atlas EZO

Pin,Unit,Type,Name,Side
1,1,gnd,GND,top
2,1,out,TX,top
3,1,in,RX,top
4,1,pwr,VCC,bottom
5,1,in,P+,bottom
6,1,in,P-,bottom

`

bullitbd, 2018-02-19:
this was caused by a blank line at the end of the file... as the instructions suggested.

leoheck, 2018-07-07:
Is it possible to ignore blank lines?
Sometimes it is good to have empty lines to organize things.

Another thing, is it possible to create a command for comments? Maybe `#` at the beginning of the line to ignore the whole line will be a good thing.

Also, I generally use text editor to do things instead of spreadsheets (they are easier to be fast)
And I like to keep comments inside the file to organize things.

bullitbd, 2018-07-08:
the spreadsheet point is well taken... really no need  to use one.

closed 2018-07-08 by bullitbd