devbisme/KiPart #45: Error when running kipart

deepakk2, 2020-01-27:
I tried running kpart using the example csv file provided in the test directory. 
I get the following error everytime I run it:

kipart example1.csv -o example1.lib
usage: kipart [-h] [-v]
              [-r [{generic,xilinxultra,xilinx7,xilinx6s,xilinx6v,psoc5lp,stm32cube,lattice,gowin}]]
              [-s [{row,num,name}]] [--reverse]
              [--side [{left,right,top,bottom}]] [-o [file.lib]] [-f] [-b]
              [-a] [-w] [-d [LEVEL]]
              file.[csv|txt|xlsx|zip] [file.[csv|txt|xlsx|zip] ...]
kipart: error: argument -r/--reader: invalid lower value: 'generic'

The version of kipart is 0.1.36

Hope I can get some feedback on this!

xesscorp, 2020-01-28:
I installed the latest `kipart` on Windows 10 and lubuntu 19 (5.3.0) and tested it with Python 3.7. I didn't encounter any errors.

xesscorp, 2020-01-30:
Do you have any other information on this error, or should I close this issue?

deepakk2, 2020-02-09:
Apologies for not replying soon. It was the python version.
 I was using Python 3.5 on both Ubuntu and Windows 10. Updated python and everything works!

Just another note, I would like to add. The version of KiPart 0.1.33 worked fine with python 3.5.

xesscorp, 2020-02-11:
I stepped all the way back to Python 3.5.3 and couldn't recreate the error. I'll leave this issue open for a while.

deepakk2, 2020-02-11:
I had python version 3.5.2 on Ubuntu. On Windows I had python 2.7.14. Not sure what exactly caused the issue, but the updated python 3.5.7 on ubuntu 19.10 worked.

closed 2020-05-12 by xesscorp