devbisme/KiPart #46: Kipart cannot import STM32CubeMX pinouts anymore?

korken89, 2020-02-24:
Hi,

It seems that some update has been made to the pinout format, when running on the attached generated CSV the following error is given:
```console
$ kipart -r stm32cube kipart.csv                                                                                                                                                                                             

Traceback (most recent call last):
  File "/usr/bin/kipart", line 11, in <module>
    load_entry_point('kipart==0.1.36', 'console_scripts', 'kipart')()
  File "/usr/lib/python3.8/site-packages/kipart/kipart.py", line 1033, in main
    call_kipart(part_data_file, input_file, file_ext)
  File "/usr/lib/python3.8/site-packages/kipart/kipart.py", line 965, in call_kipart
    return kipart(
  File "/usr/lib/python3.8/site-packages/kipart/kipart.py", line 805, in kipart
    ) in part_reader(part_data_file, part_data_file_name, part_data_file_type):
TypeError: stm32cube_reader() takes from 1 to 2 positional arguments but 3 were given
```

Tested with kipart 0.1.36 on Arch with Python 3.8.1, pinout CSV generated with STM32CubeMX 5.5.0

STM32CubeMX generated pinout CSV file:
```
"Position","Name","Type","Signal","Label","AF0","AF1","AF2","AF3","AF4","AF5","AF6","AF7","AF8","AF9","AF10","AF11","AF12","AF13","AF14","AF15"
"1","VBAT","Power","","","","","","","","","","","","","","","","","",""
"2","PC13","I/O","","","","","","","","","","","","","","","","","","EVENTOUT"
"3","PC14-OSC32_IN (PC14)","I/O","RCC_OSC32_IN","","","","","","","","","","","","","","","","","EVENTOUT"
"4","PC15-OSC32_OUT (PC15)","I/O","","","","","","","","","","","","","","","","","","EVENTOUT"
"5","PH0-OSC_IN (PH0)","I/O","","","","","","","","","","","","","","","","","","EVENTOUT"
"6","PH1-OSC_OUT (PH1)","I/O","","","","","","","","","","","","","","","","","","EVENTOUT"
"7","NRST","Reset","","","","","","","","","","","","","","","","","",""
"8","VSSA/VREF-","Power","","","","","","","","","","","","","","","","","",""
"9","VDDA/VREF+","Power","","","","","","","","","","","","","","","","","",""
"10","PA0","I/O","ADC1_IN5","","","TIM2_CH1","","","","","","USART2_CTS","","","","","COMP1_OUT","","TIM2_ETR","EVENTOUT"
"11","PA1","I/O","USART2_DE","","","TIM2_CH2","","","I2C1_SMBA","SPI1_SCK","","USART2_DE/USART2_RTS","","","","","","","TIM15_CH1N","EVENTOUT"
"12","PA2","I/O","USART2_TX","","","TIM2_CH3","","","","","","USART2_TX","LPUART1_TX","","QUADSPI_BK1_NCS","","","","TIM15_CH1","EVENTOUT"
"13","PA3","I/O","USART2_RX","","","TIM2_CH4","","","","","","USART2_RX","LPUART1_RX","","QUADSPI_CLK","","","","TIM15_CH2","EVENTOUT"
"14","PA4","I/O","SPI1_NSS","","","","","","","SPI1_NSS","","USART2_CK","","","","","","","LPTIM2_OUT","EVENTOUT"
"15","PA5","I/O","SPI1_SCK","","","TIM2_CH1","TIM2_ETR","","","SPI1_SCK","","","","","","","","","LPTIM2_ETR","EVENTOUT"
"16","PA6","I/O","SPI1_MISO","","","TIM1_BKIN","","","","SPI1_MISO","COMP1_OUT","USART3_CTS","LPUART1_CTS","","QUADSPI_BK1_IO3","","","","TIM16_CH1","EVENTOUT"
"17","PA7","I/O","SPI1_MOSI","","","TIM1_CH1N","","","I2C3_SCL","SPI1_MOSI","","","","","QUADSPI_BK1_IO2","","","","","EVENTOUT"
"18","PB0","I/O","","","SYS_TRACED0","TIM1_CH2N","","","","SPI1_NSS","","USART3_CK","","","QUADSPI_BK1_IO1","","COMP1_OUT","","","EVENTOUT"
"19","PB1","I/O","","","SYS_TRACED1","TIM1_CH3N","","","","","","USART3_DE/USART3_RTS","LPUART1_DE/LPUART1_RTS","","QUADSPI_BK1_IO0","","","","LPTIM2_IN1","EVENTOUT"
"20","PB2","I/O","","","RTC_OUT_ALARM/RTC_OUT_CALIB","LPTIM1_OUT","","","I2C3_SMBA","","","","","","","","","","","EVENTOUT"
"21","PB10","I/O","","","","TIM2_CH3","","","I2C2_SCL","SPI2_SCK","","USART3_TX","LPUART1_RX","TSC_SYNC","QUADSPI_CLK","","COMP1_OUT","","","EVENTOUT"
"22","PB11","I/O","","","","TIM2_CH4","","","I2C2_SDA","","","USART3_RX","LPUART1_TX","","QUADSPI_BK1_NCS","","","","","EVENTOUT"
"23","VSS","Power","","","","","","","","","","","","","","","","","",""
"24","VDD","Power","","","","","","","","","","","","","","","","","",""
"25","PB12","I/O","SPI2_NSS","","","TIM1_BKIN","","","I2C2_SMBA","SPI2_NSS","","USART3_CK","LPUART1_DE/LPUART1_RTS","TSC_G1_IO1","","","","","TIM15_BKIN","EVENTOUT"
"26","PB13","I/O","SPI2_SCK","","","TIM1_CH1N","","","I2C2_SCL","SPI2_SCK","","USART3_CTS","LPUART1_CTS","TSC_G1_IO2","","","","","TIM15_CH1N","EVENTOUT"
"27","PB14","I/O","SPI2_MISO","","","TIM1_CH2N","","","I2C2_SDA","SPI2_MISO","","USART3_DE/USART3_RTS","","TSC_G1_IO3","","","","","TIM15_CH1","EVENTOUT"
"28","PB15","I/O","SPI2_MOSI","","RTC_REFIN","TIM1_CH3N","","","","SPI2_MOSI","","","","TSC_G1_IO4","","","","","TIM15_CH2","EVENTOUT"
"29","PA8","I/O","TIM1_CH1","","RCC_MCO","TIM1_CH1","","","","","","USART1_CK","","","","","","","LPTIM2_OUT","EVENTOUT"
"30","PA9","I/O","TIM1_CH2","","","TIM1_CH2","","","I2C1_SCL","","","USART1_TX","","","","","","","TIM15_BKIN","EVENTOUT"
"31","PA10","I/O","TIM1_CH3","","","TIM1_CH3","","","I2C1_SDA","","","USART1_RX","","","CRS_SYNC","","","","","EVENTOUT"
"32","PA11","I/O","","","","TIM1_CH4","TIM1_BKIN2","","","SPI1_MISO","COMP1_OUT","USART1_CTS","","","USB_DM","","TIM1_BKIN2_COMP1","","","EVENTOUT"
"33","PA12","I/O","USART1_DE","","","TIM1_ETR","","","","SPI1_MOSI","","USART1_DE/USART1_RTS","","","USB_DP","","","","","EVENTOUT"
"34","PA13 (JTMS/SWDIO)","I/O","SYS_JTMS-SWDIO","","SYS_JTMS-SWDIO","IR_OUT","","","","","","","","","USB_NOE","","","","","EVENTOUT"
"35","VSS","Power","","","","","","","","","","","","","","","","","",""
"36","VDDUSB","Power","","","","","","","","","","","","","","","","","",""
"37","PA14 (JTCK/SWCLK)","I/O","SYS_JTCK-SWCLK","","SYS_JTCK-SWCLK","LPTIM1_OUT","","","I2C1_SMBA","","","","","","","","","","","EVENTOUT"
"38","PA15 (JTDI)","I/O","","","SYS_JTDI","TIM2_CH1","TIM2_ETR","USART2_RX","","SPI1_NSS","","USART3_DE/USART3_RTS","","","","","","","","EVENTOUT"
"39","PB3 (JTDO/TRACESWO)","I/O","SYS_JTDO-SWO","","SYS_JTDO-SWO","TIM2_CH2","","","","SPI1_SCK","","USART1_DE/USART1_RTS","","","","","","","","EVENTOUT"
"40","PB4 (NJTRST)","I/O","","","SYS_JTRST","","","","I2C3_SDA","SPI1_MISO","","USART1_CTS","","TSC_G2_IO1","","","","","","EVENTOUT"
"41","PB5","I/O","","","SYS_TRACED2","LPTIM1_IN1","","","I2C1_SMBA","SPI1_MOSI","","USART1_CK","","TSC_G2_IO2","","","","","TIM16_BKIN","EVENTOUT"
"42","PB6","I/O","USART1_TX","","SYS_TRACED3","LPTIM1_ETR","","","I2C1_SCL","","","USART1_TX","","TSC_G2_IO3","","","","","TIM16_CH1N","EVENTOUT"
"43","PB7","I/O","USART1_RX","","SYS_TRACECLK","LPTIM1_IN2","","","I2C1_SDA","","","USART1_RX","","TSC_G2_IO4","","","","","","EVENTOUT"
"44","PH3-BOOT0 (BOOT0)","I/O","","","","","","","","","","","","","","","","","","EVENTOUT"
"45","PB8","I/O","","","","","","","I2C1_SCL","","","","","","","","","","TIM16_CH1","EVENTOUT"
"46","PB9","I/O","","","","IR_OUT","","","I2C1_SDA","SPI2_NSS","","","","","","","","","","EVENTOUT"
"47","VSS","Power","","","","","","","","","","","","","","","","","",""
"48","VDD","Power","","","","","","","","","","","","","","","","","",""
```

korken89, 2020-02-24:
It seems that `stm32cube_reader.py` takes to few arguments.
Adding an unused argument in the middle as:

```python
def stm32cube_reader(part_data_file, unused, part_data_file_type=".csv"):
```

fixes the issue.

xesscorp, 2020-02-25:
Thanks for the report and diagnosis! I've updated the repo and released a [new version (0.1.39) on PyPi](https://pypi.org/project/kipart/).

closed 2020-02-25 by xesscorp