devbisme/KiPart #49: kipart is currently failing to build zipped files for ecp5

mattvenn, 2020-07-15:
with the lattice dir of https://github.com/xesscorp/KiCad-Schematic-Symbol-Libraries
cd lattice
make
```
kipart ECP5_pinouts.zip -r lattice -b -s name -o Lattice_ECP5_FPGA.lib -w
Traceback (most recent call last):
  File "/home/matt/.local/bin/kipart", line 8, in <module>
    sys.exit(main())
  File "/home/matt/.local/lib/python3.6/site-packages/kipart/kipart.py", line 1037, in main
    part_data_file, zipped_file.filename, zip_file_ext
  File "/home/matt/.local/lib/python3.6/site-packages/kipart/kipart.py", line 991, in call_kipart
    debug_level=args.debug,
  File "/home/matt/.local/lib/python3.6/site-packages/kipart/kipart.py", line 807, in kipart
    ) in part_reader(part_data_file, part_data_file_name, part_data_file_type):
  File "/home/matt/.local/lib/python3.6/site-packages/kipart/lattice_reader.py", line 60, in lattice_reader
    pos = csv_file.tell()
io.UnsupportedOperation: underlying stream is not seekable
makefile:14: recipe for target 'Lattice_ECP5_FPGA.lib' failed
make: *** [Lattice_ECP5_FPGA.lib] Error 1
```

This is with KiPart 0.1.43 and I have tried python2 and python3.

mattvenn, 2020-07-15:
I was still able to create the part by unzipping the zip file and making the symbol I wanted individually.

xesscorp, 2020-07-21:
I wasn't able to recreate this problem on my machine running 0.1.43/Python 3.7/Ubuntu 20.04. Is there any other information on your environment that might be pertinent?

mattvenn, 2020-07-21:
I'm on ubuntu 18.04.
Python is python3.6

I can't think of anything else. Maybe the zipfile module for 3.6 doesn't allow seeking? Let me know if you want a specific test running.

xesscorp, 2020-07-28:
Strangely enough, this seems to be a problem with Python 2.7, 3.5 and 3.6. It works OK on Python 3.7 and 3.8.

mattvenn, 2020-07-28:
I confirm it works with python3.7 on my machine.

closed 2020-11-20 by xesscorp