devbisme/KiPart #50: stm32cube reader failing in 1.43

korken89, 2020-07-19:
Hi, it seems there has been a regression/new format for stm32cube, when running now one gets the following error:

```
Traceback (most recent call last):
  File "/usr/bin/kipart", line 33, in <module>
    sys.exit(load_entry_point('kipart==0.1.43', 'console_scripts', 'kipart')())
  File "/usr/lib/python3.8/site-packages/kipart/kipart.py", line 1050, in main
    call_kipart(part_data_file, input_file, file_ext)
  File "/usr/lib/python3.8/site-packages/kipart/kipart.py", line 979, in call_kipart
    return kipart(
  File "/usr/lib/python3.8/site-packages/kipart/kipart.py", line 799, in kipart
    for (
  File "/usr/lib/python3.8/site-packages/kipart/stm32cube_reader.py", line 146, in stm32cube_reader
    ports = group_pins(pins)
  File "/usr/lib/python3.8/site-packages/kipart/stm32cube_reader.py", line 108, in group_pins
    ports[port] = sorted(ports[port], key=lambda p: parse_portpin(p[1])[1])
  File "/usr/lib/python3.8/site-packages/kipart/stm32cube_reader.py", line 108, in <lambda>
    ports[port] = sorted(ports[port], key=lambda p: parse_portpin(p[1])[1])
TypeError: 'NoneType' object is not subscriptable
```

With the following CSV:

```csv
"Position","Name","Type","Signal","Label"
"A1","PE3","I/O","",""
"A2","PE2","Output","GPIO_Output","T_VCC_EN"
"A3","PE1","I/O","",""
"A4","PE0","Output","GPIO_Output","LED3"
"A5","PB8","Output","GPIO_Output","LED2"
"A6","PB5","I/O","SPI1_MOSI",""
"A7","PG14","Output","GPIO_Output","GNDDet"
"A8","PG13","Output","GPIO_Output","RESET"
"A9","PB4","I/O","SPI1_MISO",""
"A10","PB3","I/O","SPI1_SCK",""
"A11","PD7","I/O","",""
"A12","PC12","I/O","",""
"A13","PA15","I/O","",""
"A14","PA14","I/O","SYS_JTCK-SWCLK",""
"A15","PA13","I/O","SYS_JTMS-SWDIO",""
"B1","PE4","I/O","",""
"B2","PE5","I/O","",""
"B3","PE6","I/O","",""
"B4","PB9","I/O","",""
"B5","PB7","I/O","USART1_RX",""
"B6","PB6","I/O","USART1_TX",""
"B7","PG15","I/O","",""
"B8","PG12","I/O","",""
"B9","PG11","I/O","",""
"B10","PG10","I/O","",""
"B11","PD6","I/O","USART2_RX",""
"B12","PD0","I/O","",""
"B13","PC11","I/O","",""
"B14","PC10","Output","GPIO_Output","LED1"
"B15","PA12","I/O","USB_OTG_FS_DP",""
"C1","VBAT","Power","",""
"C2","PI7","I/O","",""
"C3","PI6","I/O","",""
"C4","PI5","I/O","",""
"C5","VDD","Power","",""
"C6","PDR_ON","Reset","",""
"C7","VDD","Power","",""
"C8","VDDSDMMC","Power","",""
"C9","VDD","Power","",""
"C10","PG9","I/O","",""
"C11","PD5","I/O","USART2_TX",""
"C12","PD1","I/O","",""
"C13","PI3","I/O","SPI2_MOSI",""
"C14","PI2","I/O","SPI2_MISO",""
"C15","PA11","I/O","USB_OTG_FS_DM",""
"D1","PC13","I/O","",""
"D2","PI8","I/O","",""
"D3","PI9","I/O","",""
"D4","PI4","I/O","",""
"D5","VSS","Power","",""
"D6","BOOT0","Boot","",""
"D7","VSS","Power","",""
"D8","VSS","Power","",""
"D9","VSS","Power","",""
"D10","PD4","I/O","",""
"D11","PD3","I/O","",""
"D12","PD2","I/O","",""
"D13","PH15","I/O","",""
"D14","PI1","I/O","SPI2_SCK",""
"D15","PA10","I/O","",""
"E1","PC14-OSC32_IN","I/O","",""
"E2","PF0","I/O","",""
"E3","PI10","I/O","",""
"E4","PI11","I/O","",""
"E12","PH13","I/O","",""
"E13","PH14","I/O","",""
"E14","PI0","I/O","",""
"E15","PA9","I/O","USB_OTG_FS_VBUS",""
"F1","PC15-OSC32_OUT","I/O","",""
"F2","VSS","Power","",""
"F3","VDD","Power","",""
"F4","PH2","I/O","",""
"F6","VSS","Power","",""
"F7","VSS","Power","",""
"F8","VSS","Power","",""
"F9","VSS","Power","",""
"F10","VSS","Power","",""
"F12","VSS","Power","",""
"F13","VCAP_2","Power","",""
"F14","PC9","I/O","",""
"F15","PA8","I/O","",""
"G1","PH0-OSC_IN","I/O","RCC_OSC_IN",""
"G2","VSS","Power","",""
"G3","VDD","Power","",""
"G4","PH3","I/O","",""
"G6","VSS","Power","",""
"G7","VSS","Power","",""
"G8","VSS","Power","",""
"G9","VSS","Power","",""
"G10","VSS","Power","",""
"G12","VSS","Power","",""
"G13","VDD","Power","",""
"G14","PC8","I/O","",""
"G15","PC7","I/O","",""
"H1","PH1-OSC_OUT","I/O","RCC_OSC_OUT",""
"H2","PF2","I/O","",""
"H3","PF1","I/O","",""
"H4","PH4","I/O","",""
"H6","VSS","Power","",""
"H7","VSS","Power","",""
"H8","VSS","Power","",""
"H9","VSS","Power","",""
"H10","VSS","Power","",""
"H12","VSS","Power","",""
"H13","VDDUSB","Power","",""
"H14","PG8","I/O","",""
"H15","PC6","I/O","",""
"J1","NRST","Reset","",""
"J2","PF3","I/O","",""
"J3","PF4","I/O","",""
"J4","PH5","I/O","",""
"J6","VSS","Power","",""
"J7","VSS","Power","",""
"J8","VSS","Power","",""
"J9","VSS","Power","",""
"J10","VSS","Power","",""
"J12","VDD","Power","",""
"J13","VDD","Power","",""
"J14","V12PHYHS","Power","",""
"J15","REXTPHYHS","MonoIO","USB_OTG_HS_REXTPHYHS",""
"K1","PF7","I/O","",""
"K2","PF6","I/O","",""
"K3","PF5","I/O","",""
"K4","VDD","Power","",""
"K6","VSS","Power","",""
"K7","VSS","Power","",""
"K8","VSS","Power","",""
"K9","VSS","Power","",""
"K10","VSS","Power","",""
"K12","PH12","I/O","",""
"K13","PG5","I/O","",""
"K14","PG4","I/O","",""
"K15","PG3","I/O","",""
"L1","PF10","I/O","",""
"L2","PF9","I/O","",""
"L3","PF8","I/O","",""
"L4","BYPASS_REG","Reset","",""
"L12","PH11","I/O","",""
"L13","PH10","I/O","",""
"L14","PD15","I/O","",""
"L15","PG2","I/O","",""
"M1","VSSA","Power","",""
"M2","PC0","I/O","",""
"M3","PC1","I/O","",""
"M4","PC2","I/O","",""
"M5","PC3","I/O","",""
"M6","PB2","I/O","",""
"M7","PG1","I/O","",""
"M8","VSS","Power","",""
"M9","VSS","Power","",""
"M10","VCAP_1","Power","",""
"M11","PH6","I/O","",""
"M12","PH8","I/O","",""
"M13","PH9","I/O","",""
"M14","PD14","I/O","",""
"M15","PD13","I/O","",""
"N1","VREF-","Power","",""
"N2","PA1","I/O","",""
"N3","PA0-WKUP","I/O","",""
"N4","PA4","I/O","",""
"N5","PC4","I/O","",""
"N6","PF13","I/O","",""
"N7","PG0","I/O","",""
"N8","VDD","Power","",""
"N9","VDD","Power","",""
"N10","VDD","Power","",""
"N11","PE13","I/O","",""
"N12","PH7","I/O","",""
"N13","PD12","I/O","",""
"N14","PD11","I/O","",""
"N15","PD10","Output","GPIO_Output","GPIO2"
"P1","VREF+","Power","",""
"P2","PA2","I/O","",""
"P3","PA6","I/O","",""
"P4","PA5","I/O","",""
"P5","PC5","I/O","",""
"P6","PF12","I/O","",""
"P7","PF15","I/O","",""
"P8","PE8","I/O","",""
"P9","PE9","I/O","",""
"P10","PE11","I/O","",""
"P11","PE14","I/O","",""
"P12","PB12","I/O","",""
"P13","PB13","I/O","USB_OTG_HS_VBUS",""
"P14","PD9","I/O","",""
"P15","PD8","Output","GPIO_Output","GPIO1"
"R1","VDDA","Power","",""
"R2","PA3","I/O","ADC1_IN3",""
"R3","PA7","I/O","",""
"R4","PB1","I/O","",""
"R5","PB0","I/O","",""
"R6","PF11","I/O","",""
"R7","PF14","I/O","",""
"R8","PE7","I/O","",""
"R9","PE10","I/O","",""
"R10","PE12","I/O","",""
"R11","PE15","I/O","",""
"R12","PB10","Output","GPIO_Output","USB_SEL"
"R13","PB11","I/O","",""
"R14","PB14","I/O","USB_OTG_HS_DM",""
"R15","PB15","I/O","USB_OTG_HS_DP",""
```

korken89, 2020-07-19:
Hi, the problem is that a new group has been added that is called `other`.

Adding "other" to the list of non-pin groups fixes the issue (line 104):

```python
def group_pins(pins):
    """Groups pins together per their port name and functions. Returns a
    dictionary of {'port': [pin]}."""
    ports = defaultdict(list)

    power_names = ["VDD", "VSS", "VCAP", "VBAT", "VREF", "V12PHYHS"]
    config_names = ["RCC_OSC", "NRST", "PDR", "SWCLK", "SWDIO", "BOOT"]

    for pin in pins:
        number, name, ptype = pin
        if any(pn in name for pn in power_names):
            ports["power"].append(pin)

        elif any(pn in name for pn in config_names):
            ports["config"].append(pin)

        else:
            m = parse_portpin(name)
            if m:
                port_name, port_number = m
                ports[port_name].append(pin)
            else:
                ports["other"].append(pin)

    # sort pins
    for port in ports:
        # config and power gates are sorted according to their function name
        if port in ["config", "power", "other"]:
            ports[port] = sorted(ports[port], key=itemgetter(1))
        # IO ports are sorted according to port number
        else:
            ports[port] = sorted(ports[port], key=lambda p: parse_portpin(p[1])[1])

    return ports

```

xesscorp, 2020-07-21:
I made the change and released KiPart 0.1.44.

closed 2020-07-21 by xesscorp