devbisme/KiPart #51: Read spice files

pepijndevos, 2020-10-03:
**Is your feature request related to a problem? Please describe.**

I have a large library of spice models, and I'm looking for a good tool to do schematic entry and simulation.
It seems eeschema is one of the more mature options out there, but it would be a lot of work to manually make symbols for this library.

**Describe the solution you'd like**

I'd like to be able to auto-generate KiCad symbols for the spice models, that have all the right attributes set to use them in simulation.

In LTSpice it is possible to right-click a spice model and auto-generate a symbol from it, it would be great to have this functionality in KiCad.

**Describe alternatives you've considered**

It would also be possible to write a separate script to generate a csv file from the spice file and then run your tool.
This allows easy editing of the symbol, but adds an extra program and extra step.

Maybe it would be possible to use https://gitlab.com/kicad/libraries/kicad-library-utils/-/tree/master/schlib/autogen instead.

**Additional context**

The library I'm trying to work with is https://github.com/google/skywater-pdk

I'd be happy to try and implement this feature, but always good to check what author of the project thinks before jumping in.

xesscorp, 2020-10-03:
You could do this by making a *reader* that parses the SPICE library file and passes a dictionary back to the main `kipart` program. You'll find existing reader files in the `kipart` subdirectory.

As an alternative to doing schematic entry and simulation in KiCad, you could instead use [SKiDL/PySpice](https://github.com/xesscorp/skidl/blob/master/examples/spice-sim-intro/spice-sim-intro.ipynb) to do SPICE simulations within Python. This would eliminate the need to convert your SPICE libraries to EESCHEMA symbols.

pepijndevos, 2020-10-03:
Thanks. I'll look into it.

What I realised though, is that this particular library has many variations of basic elements. So it might make more sense to write a script that makes, say, a copy of the existing NMOS symbol, and just changes the parameters. I'll look into this more tomorrow, and get back with what I'll end up doing.

closed 2022-11-23 by devbisme