devbisme/KiPart #52: Lattice reader get stuck indefinently

korken89, 2020-11-08:
Hi, I have found that after exporting a pin description from Lattice Diamond, kipart never completes.
I have attached the pin CSV below for debugging.

Command being run: `kipart -r lattice lattice_test.csv`

**Desktop (please complete the following information):**
 - OS Arch Linux
 - Python 3.8
 - KiPart version 0.1.44

`lattice_test.csv`:

```csv
#Pin Layout Report file generated by Lattice Diamond Version 3.11.3
#Generated at Sun Nov  8 17:53:07 2020
#DESIGN = test
#DEVICE = LFE5U-85F
#PACKAGE = CABGA381
#SPEEDGRADE = 8
Pin_Number,Pad_Name,Function,IO_Bank_Number,Signal_Name,Direction
A4,PL11A,ULC_GPLL0T_IN,7,,
A5,PL11B,ULC_GPLL0C_IN,7,,
B5,PL11C,ULC_GPLL0T_MFGOUT2,7,,
C5,PL11D,ULC_GPLL0C_MFGOUT2,7,,
C4,PL14A,ULC_GPLL0T_MFGOUT1,7,,
A3,PL14C,,7,,
B4,PL14B,ULC_GPLL0C_MFGOUT1,7,,
B3,PL14D,,7,,
E4,PL17A,,7,,
C3,PL17C,,7,,
D5,PL17B,,7,,
D3,PL17D,,7,,
F4,PL20A,,7,,
E5,PL20C,,7,,
E3,PL20B,,7,,
F5,PL20D,,7,,
A2,PL35A,,7,,
B2,PL35C,VREF1_7,7,,
B1,PL35B,,7,,
C2,PL35D,,7,,
C1,PL38A,,7,,
D2,PL38C,,7,,
D1,PL38B,,7,,
E1,PL38D,,7,,
H4,PL41A,GR_PCLK7_1,7,,
H5,PL41C,GR_PCLK7_0,7,,
G5,PL41B,,7,,
H3,PL41D,,7,,
G3,PL44A,PCLKT7_1,7,,
F2,PL44C,PCLKT7_0,7,,
F3,PL44B,PCLKC7_1,7,,
E2,PL44D,PCLKC7_0,7,,
G2,PL47A,PCLKT6_1,6,,
H2,PL47C,PCLKT6_0,6,,
F1,PL47B,PCLKC6_1,6,,
G1,PL47D,PCLKC6_0,6,,
J4,PL50A,GR_PCLK6_0,6,,
J3,PL50C,GR_PCLK6_1,6,,
J5,PL50B,,6,,
K3,PL50D,,6,,
K2,PL53A,,6,,
H1,PL53C,,6,,
J1,PL53B,,6,,
K1,PL53D,,6,,
K4,PL56A,,6,,
L4,PL56C,,6,,
K5,PL56B,VREF1_6,6,,
L5,PL56D,,6,,
M5,PL77A,,6,,
M4,PL83A,,6,,
N4,PL83C,D15/IO15,6,,
N5,PL83B,,6,,
P5,PL83D,D14/IO14,6,,
N3,PL86A,D13/IO13,6,,
L3,PL86C,D11/IO11,6,,
M3,PL86B,D12/IO12,6,,
L2,PL86D,D10/IO10,6,,
N2,PL89A,LLC_GPLL0T_MFGOUT2,6,,
L1,PL89C,D9/IO9,6,,
M1,PL89B,LLC_GPLL0C_MFGOUT2,6,,
N1,PL89D,D8/IO8,6,,
P1,PL92A,LLC_GPLL0T_MFGOUT1,6,,
P2,PL92B,LLC_GPLL0C_MFGOUT1,6,,
P3,PL92C,LLC_GPLL0T_IN,6,,
P4,PL92D,LLC_GPLL0C_IN,6,,
R1,PB4A,D7/IO7,8,,
U1,PB6A,D5/MISO2/IO5,8,,
T1,PB4B,D6/IO6,8,,
V1,PB6B,D4/MOSI2/IO4,8,,
W1,PB9A,D3/IO3,8,,
V2,PB11A,D1/MISO/IO1,8,,
Y2,PB9B,D2/IO2,8,,
W2,PB11B,D0/MOSI/IO0,8,,
T2,PB13A,SN/CSN/SCAN_SHFT_EN,8,,
R2,PB15A,HOLDN/DI/BUSY/CSSPIN/CEN,8,,
U2,PB13B,CS1N,8,,
R3,PB15B,DOUT/CSON/ATB_FORCE,8,,
T3,PB18A,WRITEN/ATB_SENSE,8,,
V3,INITN,,8,,
U3,CCLK,MCLK/SCK,8,,
W3,PROGRAMN,,8,,
Y3,DONE,,8,,
T4,CFG_1,,8,,
R4,CFG_2,,8,,
U4,CFG_0,,8,,
V4,TDO,,,,
T5,TCK,TEST_CLK,,,
R5,TDI,,,,
U5,TMS,,,,
W4,RESERVED,,,,
T7,GND,,,,
W5,RESERVED,,,,
Y5,GND,,,,
T8,GND,,,,
Y6,GND,,,,
Y7,GND,,,,
T9,GND,,,,
Y8,GND,,,,
W8,RESERVED,,,,
T10,GND,,,,
W9,RESERVED,,,,
W10,RESERVED,,,,
W11,RESERVED,,,,
Y11,GND,,,,
Y12,GND,,,,
W13,RESERVED,,,,
T11,GND,,,,
W14,RESERVED,,,,
Y14,GND,,,,
T12,GND,,,,
Y15,GND,,,,
Y16,GND,,,,
T13,GND,,,,
Y17,GND,,,,
W17,RESERVED,,,,
T14,GND,,,,
W18,RESERVED,,,,
Y19,GND,,,,
W20,GND,,,,
T17,PR92D,LRC_GPLL0C_IN/S7_OUT,3,,
U16,PR92C,LRC_GPLL0T_IN,3,,
U17,PR92B,LRC_GPLL0C_MFGOUT1,3,,
U18,PR92A,LRC_GPLL0T_MFGOUT1,3,,
T18,PR89D,LRC_GPLL0C_MFGOUT2,3,,
R18,PR89B,,3,,
U19,PR89C,LRC_GPLL0T_MFGOUT2,3,,
T19,PR89A,,3,,
U20,PR86D,,3,,
R20,PR86B,S7_IN,3,,
T20,PR86C,,3,,
P20,PR86A,S6_OUT,3,,
P18,PR83D,,3,,
N20,PR83B,S6_IN,3,,
P19,PR83C,,3,,
N19,PR83A,S5_OUT,3,,
T16,PR77A,,3,,
R17,PR56D,,3,,
P16,PR56B,VREF1_3,3,,
R16,PR56C,S5_IN,3,,
N17,PR56A,S4_OUT,3,,
P17,PR53D,,3,,
M17,PR53B,,3,,
N18,PR53C,,3,,
N16,PR53A,,3,,
M18,PR50D,,3,,
L17,PR50B,,3,,
L18,PR50C,GR_PCLK3_1,3,,
L16,PR50A,GR_PCLK3_0,3,,
M19,PR47D,PCLKC3_0,3,,
M20,PR47B,PCLKC3_1,3,,
L19,PR47C,PCLKT3_0/S4_IN,3,,
L20,PR47A,PCLKT3_1/S3_OUT,3,,
K20,PR44D,PCLKC2_0,2,,
K19,PR44B,PCLKC2_1,2,,
J20,PR44C,PCLKT2_0/S3_IN,2,,
J19,PR44A,PCLKT2_1/S2_OUT,2,,
K18,PR41D,,2,,
H20,PR41B,,2,,
J18,PR41C,GR_PCLK2_0,2,,
G19,PR41A,GR_PCLK2_1,2,,
G20,PR38D,,2,,
F19,PR38B,,2,,
F20,PR38C,,2,,
E20,PR38A,,2,,
E19,PR35D,,2,,
D19,PR35B,S2_IN,2,,
D20,PR35C,VREF1_2,2,,
C20,PR35A,S1_OUT,2,,
K17,PR29B,,2,,
K16,PR29A,,2,,
J16,PR20D,,2,,
H17,PR20B,,2,,
J17,PR20C,,2,,
H18,PR20A,,2,,
H16,PR17D,,2,,
G18,PR17B,URC_GPLL0C_MFGOUT1,2,,
G16,PR17C,,2,,
F17,PR17A,URC_GPLL0T_MFGOUT1,2,,
F18,PR14D,,2,,
E17,PR14B,S1_IN,2,,
E18,PR14C,,2,,
D18,PR14A,S0_OUT,2,,
F16,PR11D,URC_GPLL0C_MFGOUT2,2,,
E16,PR11C,URC_GPLL0T_MFGOUT2,2,,
D17,PR11B,URC_GPLL0C_IN/S0_IN,2,,
C18,PR11A,URC_GPLL0T_IN,2,,
B20,PT121B,URC_GPLL1C_IN,1,,
B19,PT119B,,1,,
A19,PT121A,URC_GPLL1T_IN,1,,
A18,PT119A,,1,,
B18,PT116B,,1,,
C17,PT114B,,1,,
A17,PT116A,,1,,
B17,PT114A,,1,,
D16,PT112B,,1,,
B16,PT110B,,1,,
C16,PT112A,,1,,
A16,PT110A,,1,,
E15,PT107B,,1,,
C15,PT105B,,1,,
D15,PT107A,,1,,
B15,PT105A,,1,,
A15,PT103A,,1,,
E14,PT85B,,1,,
C14,PT83B,,1,,
D14,PT85A,,1,,
A14,PT83A,,1,,
E13,PT80B,,1,,
C13,PT78B,,1,,
D13,PT80A,,1,,
B13,PT78A,,1,,
A13,PT76B,,1,,
E12,PT74B,GR_PCLK1_1,1,,
A12,PT76A,,1,,
D12,PT74A,GR_PCLK1_0,1,,
C12,PT71B,,1,,
E11,PT69B,,1,,
B12,PT71A,PCLKT1_0,1,,
D11,PT69A,PCLKT1_1,1,,
C11,PT65B,,0,,
A11,PT63B,,0,,
B11,PT65A,PCLKT0_0,0,,
A10,PT63A,PCLKT0_1,0,,
B10,PT60B,GR_PCLK0_0,0,,
C10,PT58B,,0,,
A9,PT60A,GR_PCLK0_1,0,,
B9,PT58A,,0,,
E10,PT56B,,0,,
D10,PT56A,,0,,
C9,PT54A,,0,,
E9,PT20B,,0,,
A8,PT18B,,0,,
D9,PT20A,,0,,
A7,PT18A,,0,,
B8,PT15B,,0,,
D8,PT13B,,0,,
C8,PT15A,,0,,
E8,PT13A,,0,,
C7,PT11B,,0,,
D7,PT9B,,0,,
C6,PT11A,,0,,
E7,PT9A,,0,,
D6,PT6B,,0,,
B6,PT4B,ULC_GPLL1C_IN,0,,
E6,PT6A,,0,,
A6,PT4A,ULC_GPLL1T_IN,0,,
C19,GND,,,,Gnd
H19,GND,,,,Gnd
R19,GND,,,,Gnd
G17,GND,,,,Gnd
M16,GND,,,,Gnd
G15,GND,,,,Gnd
K15,GND,,,,Gnd
N15,GND,,,,Gnd
B14,GND,,,,Gnd
F14,GND,,,,Gnd
G14,GND,,,,Gnd
J14,GND,,,,Gnd
K14,GND,,,,Gnd
M14,GND,,,,Gnd
N14,GND,,,,Gnd
P14,GND,,,,Gnd
F13,GND,,,,Gnd
G13,GND,,,,Gnd
P13,GND,,,,Gnd
G12,GND,,,,Gnd
J12,GND,,,,Gnd
K12,GND,,,,Gnd
L12,GND,,,,Gnd
M12,GND,,,,Gnd
P12,GND,,,,Gnd
G11,GND,,,,Gnd
J11,GND,,,,Gnd
K11,GND,,,,Gnd
L11,GND,,,,Gnd
M11,GND,,,,Gnd
P11,GND,,,,Gnd
G10,GND,,,,Gnd
J10,GND,,,,Gnd
K10,GND,,,,Gnd
L10,GND,,,,Gnd
M10,GND,,,,Gnd
G9,GND,,,,Gnd
J9,GND,,,,Gnd
K9,GND,,,,Gnd
L9,GND,,,,Gnd
M9,GND,,,,Gnd
F8,GND,,,,Gnd
G8,GND,,,,Gnd
P8,GND,,,,Gnd
B7,GND,,,,Gnd
F7,GND,,,,Gnd
G7,GND,,,,Gnd
J7,GND,,,,Gnd
K7,GND,,,,Gnd
M7,GND,,,,Gnd
N7,GND,,,,Gnd
P7,GND,,,,Gnd
G6,GND,,,,Gnd
K6,GND,,,,Gnd
N6,GND,,,,Gnd
D4,GND,,,,Gnd
G4,GND,,,,Gnd
J2,GND,,,,Gnd
M2,GND,,,,Gnd
H13,VCC,,,,Power
J13,VCC,,,,Power
K13,VCC,,,,Power
L13,VCC,,,,Power
M13,VCC,,,,Power
N13,VCC,,,,Power
H12,VCC,,,,Power
N12,VCC,,,,Power
H11,VCC,,,,Power
N11,VCC,,,,Power
H10,VCC,,,,Power
N10,VCC,,,,Power
H9,VCC,,,,Power
N9,VCC,,,,Power
H8,VCC,,,,Power
J8,VCC,,,,Power
K8,VCC,,,,Power
L8,VCC,,,,Power
M8,VCC,,,,Power
N8,VCC,,,,Power
F15,VCCAUX,,,,
P15,VCCAUX,,,,
F6,VCCAUX,,,,
P6,VCCAUX,,,,
F10,VCCIO0,,0,,
F9,VCCIO0,,0,,
F12,VCCIO1,,1,,
F11,VCCIO1,,1,,
H15,VCCIO2,,2,,
J15,VCCIO2,,2,,
H14,VCCIO2,,2,,
L15,VCCIO3,,3,,
M15,VCCIO3,,3,,
L14,VCCIO3,,3,,
V20,GND,,,,
V19,GND,,,,
W19,GND,,,,
V16,GND,,,,
W16,GND,,,,
V15,GND,,,,
W15,GND,,,,
U14,GND,,,,
V14,GND,,,,
U13,GND,,,,
V13,GND,,,,
U12,GND,,,,
V12,GND,,,,
W12,GND,,,,
U11,GND,,,,
U10,GND,,,,
U9,GND,,,,
V9,GND,,,,
U8,GND,,,,
V8,GND,,,,
U7,GND,,,,
V7,GND,,,,
W7,GND,,,,
V6,GND,,,,
W6,GND,,,,
V5,GND,,,,
T6,VCCA0,,,,
U6,VCCA0,,,,
T15,VCCA1,,,,
U15,VCCA1,,,,
V11,VCCAUXA0,,,,
V10,VCCAUXA0,,,,
V18,VCCAUXA1,,,,
V17,VCCAUXA1,,,,
P10,VCCIO8,,8,,
P9,VCCIO8,,8,,
L7,VCCIO6,,6,,
L6,VCCIO6,,6,,
M6,VCCIO6,,6,,
H7,VCCIO7,,7,,
H6,VCCIO7,,7,,
J6,VCCIO7,,7,,
```

xesscorp, 2020-11-10:
I don't think KiPart works with the output of Diamond any more (at least, not the Diamond output you sent). The Diamond output is already a CSV file, so your best option right now is to manually edit the file, remove any columns you don't need (you'll probably want to keep Pin_Number, IO_Bank_Number, Signal_Name, Direction), and rename those columns to those used in a generic KiPart input file. Then pass that file through KiPart to get your Lattice symbol.

korken89, 2020-11-11:
Ah, I see. I must have misunderstood the docs.
I found a zip file in the repo that had Lattice parts, I used that one to generate a Lattice library which worked great!

Thanks for the help!

closed 2020-11-11 by korken89

xesscorp, 2020-11-11:
No, you understood the docs correctly. They do mention the support for files generated by Diamond. But over time, that feature must have been lost. I'll update the docs.

korken89, 2020-11-11:
Thanks for the clarification!