devbisme/KiPart #55: [KiPart BUG] Not a bug. KiPart is AWESOME!

adamfeuer, 2020-11-18:
**Describe the bug**
Not a bug. KiPart is AWESOME!

**Additional context**

I had a very complex library for a microprocessor that I needed to refactor— to reorder hundreds of pins in a sane human-usable format. I thought I was going to have do it by hand in the graphical symbol editor. I tried that, got very frustrated with all the clicking and dragging, searched and found this tool, used `kilib2csv` to export my KiCAD lib to CSV, edited the CSV in a spreadsheet program, exported back to CSV, used `kipart` to export back to KiCAD. Boom! Usable part in minutes.

This tool is awesome.

Thank you.

eabase, 2020-11-19:
You should add this to the README!

xesscorp, 2020-11-20:
Hi, Adam. Thanks for the note. I'm glad KiPart saved you some effort. Your story is similar to the things that made me make KiPart.

Where did you search to find KiPart? I would be interested to know because there are a lot of useful add-on tools for KiCad and I'm not sure where people find out about them (if they even do).

adamfeuer, 2020-11-20:
@xesscorp I found the library on this page: https://github.com/xesscorp/kicad-3rd-party-tools – the compilation was extremely helpful. Lots of good tools there. Thank you for that too.

adamfeuer, 2020-11-20:
@eabase https://github.com/xesscorp/KiPart/pull/56

closed 2020-11-20 by xesscorp