devbisme/KiPart #60: Option for origin in the center of the symbol

cnieves1, 2022-05-23:
Kicad library requirement S3.1 requires the origin of the symbol to be in the middle of the symbol:

https://klc.kicad.org/symbol/s3/s3.1/

I don't see any option to have the symbol centered. 

Is there any way to do this? Would be hard to add?

Thanks in advance,

cnieves1, 2022-06-13:
Hi,
I started to write the code to do this. It is currently at:
  https://github.com/cnieves1/KiPart/tree/center_symbol

I'm using it with a real CSV, attached:
[SMARC_pin_table2.csv](https://github.com/devbisme/KiPart/files/8890468/SMARC_pin_table2.csv)

I'm not quite happy with the results. I admit that I don't know KiPart internals very well, specially regarding the transforms using to rotate/translate the pins...

The pins are not quite centered in the box.

Another thing I noticed is that I would want to move text positions (datasheets, description, reference and so on) so they are outside the box, but at the time they are added to part_dfn, there is still no calculation of the box... could those be just moved at the end of the calculations?

Could you help me to polish and improve it to an state where it can be merged with master?

Thanks,

devbisme, 2022-06-27:
I think you were close! I made some changes and checked it in. Take a look and see if it does what you want.
```bash
pip install git+https://github.com/devbisme/KiPart@cnieves1
```

cnieves1, 2022-06-28:
Hi!,
Thanks a lot for taking a look at this!

The output is pretty good. I only found some small things to be improved.

In the image below, I think left and right sides should have their pins aligned, since they have the same number of pins:

[image]

In the image below, apart from using the bundle option, would be possible to reduce the Y size of the symbol? There is enough space for top and bottom pin names between pin names on the left and pin names of the right...

[image]

Thanks again!

devbisme, 2022-07-05:
I checked in a new version that fixes the alignment problem. It also adds a new option (`scrunch`) that will push the left/right pin columns under the top/bottom pin rows. Check it out and let me know.

cnieves1, 2022-07-14:
Seems perfect!
I didn't find any problem with the new version, and every symbol is now looking nice!
I guess you can now merge this branch into master.

Thanks a lot for your support!!

cnieves1, 2022-07-14:
Mmm.. I tried a merge request at kicad symbols repository and there are issues with the symbols, regarding text size and line widths...

See: https://gitlab.com/kicad/libraries/kicad-symbols/-/merge_requests/3556

It may be worth to modify kioart to stick to Kicad KLC...

Thanks

Carlos

devbisme, 2022-07-15:
I changed the box line width and font sizes. Try it and see if it fixes the problem.

cnieves1, 2022-07-15:
I get several issues:

- It is needed to set pin name offset to 20mils:

`PIN_NAME_OFFSET = 20  # Separation between pin and pin name.`

- I don't know why, but Kicad 6 is not getting right the description field:
  
[image]

The beginning of the file is:
```
EESchema-LIBRARY Version 2.3
DEF SMARC_connector J 0 20 Y Y 11 L N
F0 "J" 0 50 50 H V C CNN
F1 "SMARC_connector" 0 -50 50 H V C CNN
F2 "Connector_Amphenol:Amphenol_10151114-00xxLF_SMARC" 0 -150 50 H I C CNN
F3 "https://sget.org/standards/smarc/" 0 -350 50 H I C CNN
F5 "SMARC connector, https://sget.org/standards/smarc/" 0 -450 50 H I C CNN "desc"

```

- The bundle option does set pin number size to 0, I think. I get these errors:

```
  Violating S3.2 - https://klc.kicad.org/symbol/s3/s3.2/
    Text fields should use a common text size of 50mils
     - Pin GND (P100), text size 50, number size 0
     - Pin GND (P103), text size 50, number size 0
     - Pin GND (P12), text size 50, number size 0
     - Pin GND (P120), text size 50, number size 0
     - Pin GND (P133), text size 50, number size 0
     - Pin GND (P142), text size 50, number size 0
     - Pin VDD_IN (P147), text size 50, number size 0
     - Pin VDD_IN (P148), text size 50, number size 0
     - Pin VDD_IN (P149), text size 50, number size 0
     - Pin GND (P15), text size 50, number size 0
     - Pin VDD_IN (P150), text size 50, number size 0
     - Pin VDD_IN (P151), text size 50, number size 0
     - Pin VDD_IN (P152), text size 50, number size 0
     - Pin VDD_IN (P153), text size 50, number size 0
     - Pin VDD_IN (P154), text size 50, number size 0
     - Pin VDD_IN (P155), text size 50, number size 0
     - Pin VDD_IN (P156), text size 50, number size 0
     - Pin GND (P18), text size 50, number size 0
     - Pin GND (P2), text size 50, number size 0
     - Pin GND (P32), text size 50, number size 0
     - Pin GND (P38), text size 50, number size 0
     - Pin GND (P47), text size 50, number size 0
     - Pin GND (P50), text size 50, number size 0
     - Pin GND (P53), text size 50, number size 0
     - Pin GND (P59), text size 50, number size 0
     - Pin GND (P68), text size 50, number size 0
     - Pin GND (P79), text size 50, number size 0
     - Pin GND (P82), text size 50, number size 0
     - Pin GND (P85), text size 50, number size 0
     - Pin GND (P88), text size 50, number size 0
     - Pin GND (P9), text size 50, number size 0
     - Pin GND (P91), text size 50, number size 0
     - Pin GND (P94), text size 50, number size 0
     - Pin GND (P97), text size 50, number size 0
     - Pin GND (S10), text size 50, number size 0
     - Pin GND (S101), text size 50, number size 0
     - Pin GND (S110), text size 50, number size 0
     - Pin GND (S119), text size 50, number size 0
     - Pin GND (S124), text size 50, number size 0
     - Pin GND (S13), text size 50, number size 0
     - Pin GND (S130), text size 50, number size 0
     - Pin GND (S136), text size 50, number size 0
     - Pin GND (S143), text size 50, number size 0
     - Pin GND (S158), text size 50, number size 0
     - Pin GND (S16), text size 50, number size 0
     - Pin GND (S25), text size 50, number size 0
     - Pin GND (S3), text size 50, number size 0
     - Pin GND (S34), text size 50, number size 0
     - Pin GND (S47), text size 50, number size 0
     - Pin GND (S61), text size 50, number size 0
     - Pin GND (S64), text size 50, number size 0
     - Pin GND (S67), text size 50, number size 0
     - Pin GND (S70), text size 50, number size 0
     - Pin GND (S73), text size 50, number size 0
     - Pin GND (S80), text size 50, number size 0
     - Pin GND (S83), text size 50, number size 0
     - Pin GND (S86), text size 50, number size 0
     - Pin GND (S89), text size 50, number size 0
     - Pin GND (S92), text size 50, number size 0
```

- Did you change anything related to pin positions? I am now getting off grid pins, and I don't remember seeing that yesterday...

cnieves1, 2022-07-15:
I have double checked and the off grid pin positions were already yesterday:
  https://gitlab.com/cnieves1/kicad-symbols/-/jobs/2723281994

devbisme, 2022-07-17:
Made  commit to address these problems:
* According to the KLC, pin name offsets up to 50 mils should be acceptable, but I set it to 20 mils.
* Pin number size for bundled pins should now be set to 50.
* Pin positions for centered symbols should now be on grid.

cnieves1, 2022-07-19:
Thanks again!

I needed to set description by hand (already reported above). 

After this round, the complaints are only regarding small tweaks to symbol centering (https://gitlab.com/cnieves1/kicad-symbols/-/jobs/2742840250):
```

  Violating S3.1 - https://klc.kicad.org/symbol/s3/s3.1/
    Origin is centered on the middle of the symbol
    Symbol unit 1 slightly off-center
     -   Center calculated @ (0, -50)
    Symbol unit 3 slightly off-center
     -   Center calculated @ (50, 0)
    Symbol unit 6 slightly off-center
     -   Center calculated @ (50, -50)
    Symbol unit 7 slightly off-center
     -   Center calculated @ (0, -50)
    Symbol unit 8 slightly off-center
     -   Center calculated @ (0, -50)
    Symbol unit 9 slightly off-center
     -   Center calculated @ (-50, 0)
    Symbol unit 10 slightly off-center
     -   Center calculated @ (0, -50)
    Symbol unit 11 slightly off-center
     -   Center calculated @ (0, -50)

```
I don't know how important is this, or if the reviewer would want to fix this...

Anyway, all these changes really improved KiPart... Thanks again for your time!!

cnieves1, 2022-07-21:
I fully agree with you.

Let's see what the reviewer says...

Thanks again for your support!

devbisme, 2022-07-26:
Was your footprint accepted? If so, I'll close this issue.

cnieves1, 2022-07-26:
You mean the symbol.... 
No news from the reviewers yet..  

Anyway, the goal of this issue was to be able to have centered symbols according to KLC.

This is already accomplished, so I think this issue can safely be closed.

If there is any issue in the future, I'll get in touch again.

Thanks again for your support!!

closed 2022-07-26 by cnieves1