devbisme/KiPart #61: [KiPart BUG]: Bundle doesn't work as expected when stacking power and passive pins as per KLC S4.3

cnieves1, 2022-05-23:
Kicad library rule:

https://klc.kicad.org/symbol/s4/s4.3/

Specifies that invisible pins when stacking should be have passive pin type.

However, KiPart only bundles pins when those pins are power type.

Changing:
```
def is_pwr(pin, fuzzy_match):
    """Return true if this is a power input pin."""
    return (
        find_closest_match(name=pin.type, name_dict=PIN_TYPES, fuzzy_match=fuzzy_match)
        == "W"
    )
```

to 

```
def is_pwr(pin, fuzzy_match):
    """Return true if this is a power input pin."""
    return (
        (find_closest_match(name=pin.type, name_dict=PIN_TYPES, fuzzy_match=fuzzy_match)
        == "W") or 
        (find_closest_match(name=pin.type, name_dict=PIN_TYPES, fuzzy_match=fuzzy_match)
        == "P")
    )

```

does the trick.

dnschneid, 2022-06-08:
Does #63 do what you expect?

cnieves1, 2022-06-10:
Thanks for your answer!

It almost does it.
The only missing feature is that it appends "[n]" string, where n is the number of bundled pins.

Official Kicad library symbols don't do that. Any chance to remove that?

cnieves1, 2022-06-10:
As some people would like to keep the current behaviour, what about adding an option to do that?

```
diff --git a/kipart/kipart.py b/kipart/kipart.py
index 4b6ccef..9cc06b8 100644
--- a/kipart/kipart.py
+++ b/kipart/kipart.py
@@ -502,6 +511,8 @@ def draw_symbol(
     fill,
     box_line_width,
     push,
+    annotate_pins,
 ):
     """Add a symbol for a part to the library."""
 
@@ -618,13 +629,14 @@ def draw_symbol(
         transform = {}
 
         # Annotate the pins for each side of the symbol.
-        for side_pins in list(unit.values()):
-            annotate_pins(list(side_pins.items()))
+        if annotate_pins:
+            for side_pins in list(unit.values()):
+                annotate_pins(list(side_pins.items()))
 
         # Determine the actual bounding box for each side.
         bbox = {}
         for side, side_pins in list(unit.items()):
@@ -805,6 +818,8 @@ def kipart(
     reverse=False,
     fuzzy_match=False,
     bundle=False,
+    annotate_pins=True,
     debug_level=0,
 ):
     """Read part pin data from a CSV/text/Excel file and write or append it to a library file."""
@@ -845,6 +860,8 @@ def kipart(
             fill=fill,
             box_line_width=box_line_width,
             push=push,
+            annotate_pins=annotate_pins,
         )
 
 
@@ -891,6 +908,8 @@ def call_kipart(args, part_reader, part_data_file, file_name, file_type, parts_l
         reverse=args.reverse,
         fuzzy_match=args.fuzzy_match,
         bundle=args.bundle,
+        annotate_pins=args.annotate-pins,
         debug_level=args.debug,
     )
 
@@ -985,6 +1004,18 @@ def main():
         action="store_true",
         help="Bundle multiple, identically-named power and ground pins each into a single schematic pin.",
     )
+    parser.add_argument(
+        "-t",
+        "--annotate-pins",
+        action="store_true",
+        help="Annotate pin names if there when stacking pins'",
+    )
     parser.add_argument(
         "-a",
         "--append",

```
Thoughts?

dnschneid, 2022-06-10:
Yeah, I personally prefer `[n-1:0]` rather than `[n]`, so something like that where you can pick 0, 1, or 2 would mean I wouldn't have to maintain my own patchset to get the format I want :)

cnieves1, 2022-06-10:
Mmm... I'm curious... what features where added or modified with your patchset?
I noticed another deviation from KLC, which is that symbols should be centered... do you have any patch to do this?

dnschneid, 2022-06-10:
Nope.  Mostly just adding additional fields and formatting/positioning them differently.

dnschneid, 2022-06-10:
PR #64

cnieves1, 2022-06-10:
Great!! Thanks a lot!! I will use both!
Looking forward to getting both MR merged!

devbisme, 2022-06-12:
kipart currently only works with version 5 symbol libraries. What's your current solution to using it for version 6? Just import it using KiCad6? (Or are you not using version 6?)

dnschneid, 2022-06-12:
Yeah, I just import into KiCad6.  So far the only thing I've missed (at least, among the new features that would make sense for csv-based symbol generation) is support for the "free" pin type, which is relatively rare and easy to apply afterwards.

devbisme, 2022-06-13:
You can try the updated kipart using this command:
```bash
pip install git+https://github.com/devbisme/kipart
```
Let me know if there are any problems. Then I can upload it to PyPi.

dnschneid, 2022-06-13:
Yeah, my first patch hid the pin numbers, since I figured showing one random pin out of a stack was a bit arbitrary.  I suppose, though, it makes sense to show one of the pin numbers for stacked power nets (if you need to probe one pin, you've probed them all), but I don't think it makes sense to show any pin numbers for stacked NC pins.

dnschneid, 2022-06-13:
#65 implements that.

dnschneid, 2022-11-09:
I think this can be closed now?

cnieves1, 2022-11-09:
Yes! Thanks!

closed 2022-11-09 by cnieves1