devbisme/KiPart #68: update for kicad version 6

putyn, 2022-08-22:
**First of thank you for this piece of software! creating symbols has never been easier;**

**Is your feature request related to a problem? Please describe.**
with Kicad version 6 the transition to S-Expressions has finished (now symbols/libraries are shaved as S expressions). This is not really an issue (or at least for me) for the normal functionality of kipart because importing an older symbol (.lib) still works as expected.
However if you want to use kilib2csv to convert a symbol back to a csv file, conversion doesn't fail but the resulting csv is empty.

**Describe the solution you'd like**
Ideally kipart (kilb2csv) should be updated to also handle the newer format.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

devbisme, 2022-08-25:
I updated `kilib2csv` so it can handle KiCad V6 symbol libraries. You can install it like this:
```python
pip install git+https://github.com/devbisme/KiPart@v6
```
Unfortunately, this version of KiPart only works with **Python versions 3.10 and higher** because of restrictions with the `kiutils` library that I used. I'll probably have to find a different solution, but you can try this for now.

devbisme, 2022-08-26:
I updated the `v6` branch to remove the dependency on `kiutils`. Now it should work with any version of Python.

putyn, 2022-08-27:
sweet! thanks for taking a look into this; I will try this in the next couple of days!
I have noticed that for [kinparse](https://github.com/devbisme/kinparse) you have used [pyparsing](https://pypi.python.org/pypi/pyparsing) to parse the s-expression and I have actually started to write a parser based on your work for the kicad library format, any reason why you didn't take the same approach here?

devbisme, 2022-08-27:
In my limited experience with PyParsing, I've found it really slow. Also, the parsing statements can have some non-obvious interactions and can be really hard to follow after you haven't looked at them for a while. And when an input file has a parse error, PyParsing aborts but often doesn't point out the exact location where the parsing failed. (These drawbacks may be the result of the way I'm using PyParsing. It's hard to tell if the fault is with PyParsing or me.)

The symbol data I needed from the `.kicad_sym` file was relatively easy to extract from the nested lists generated by `sexpdata`. And `sexpdata` processes files quickly, hasn't been modified in almost a decade, and is compatible with every version of Python 2 & 3 (PyParsing needs version 3.6 or newer).

mvnmgrx, 2022-09-10:
Hello, `kiutils` author here,
FYI: Support for Python versions from 3.7 to 3.10 was added in version [v1.1.4](https://github.com/mvnmgrx/kiutils/releases/tag/v1.1.4) today. But if your requirements demand support for older versions as far as Python2, this will unfortunately be outside the scope of `kituils` ...

devbisme, 2022-09-11:
Thanks for the update!

putyn, 2022-09-11:
had a look at the V6 branch, I had an issue where the symbol name was not the same as the part as the filed Value and the assertion error was thrown, after fixing the naming issue I could generate the CSV file.
Another thing I noticed is the other fields (package, footprint, description) are not exported to the CSV file

aslansq, 2024-12-15:
code offer

Hi @devbisme 
I also developed part generator, I have some un-related use case for it.
I think this could be useful to you, generating in new lib format
If you are interested with it. I can port it to this repo
https://github.com/aslansq/KiProject/blob/main/templates/v8/e5ea7ba.kicad_sym