devbisme/KiPart #70: [KiPart BUG]

wschoenfeld, 2022-10-21:
Failed to load on Mac with Python 3.10.5, seems there is a coding issue with pin_data.strip() function

Heres the what im getting on my console
Warning: Replaced whitespace with '_' in pin '		DM4_N/ DB14_N' of part 135.
Warning: Replaced whitespace with '_' in pin '183	io' of part 135.
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.10/bin/kipart", line 33, in <module>
    sys.exit(load_entry_point('kipart==1.1.0', 'console_scripts', 'kipart')())
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/kipart/kipart.py", line 1211, in main
    call_kipart(
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/kipart/kipart.py", line 967, in call_kipart
    return kipart(
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/kipart/kipart.py", line 897, in kipart
    for (
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/kipart/generic_reader.py", line 108, in generic_reader
    setattr(pin, a, fix_pin_data(row_dict[c], part_num))
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/kipart/common.py", line 149, in fix_pin_data
    fixed_pin_data = pin_data.strip()  # Remove leading/trailing spaces.
AttributeError: 'NoneType' object has no attribute 'strip'

devbisme, 2022-11-08:
I recreated your error by building a CSV file with some malformed pin fields containing spaces. You can try the modified `kipart` by installing it as follows:
```
pip install git+https://github.com/devbisme/kipart@dev
```

devbisme, 2022-11-23:
I released KiPart 1.4.0 on PyPi with a change that should fix this problem.

closed 2022-11-23 by devbisme