devbisme/KiPart #76: Append/Overwrite Flags Issue [KiPart BUG]

CMcGGitHub, 2023-01-04:
**Describe the bug**
-a (append) and -w (overwrite) flags not behaving as documented. Want to be able to overwrite **existing** parts on an **existing** library. 

Documentation says 
_-a, --append, --add              Add parts to an existing part library. **Overwrite existing parts only if used in conjunction with -w**_

**To Reproduce**
_Create test.lib containing 2 symbols_
     kipart -o test.lib -a symbol1.csv symbol2.csv     (Writing test.lib 2)

_Update **only** symbol1 in test.lib_
     kipart -o test.lib -a -w symbol1.csv  (Writing test.lib **1**)

**Expected behavior**
In above scenario, I'd expect second step to update test.lib with the updated symbol1 **and** the unchanged symbol2 however only symbol1 gets written.

**Workaround**
As I maintain all my kipart generated files in a single library, I work around this by putting all symbol.csv files onto the command line every time, instead of doing additions/modifications in an incremental manner.

**Versions**
 - OS Win10
 - Python version 3.9.14
 - KiPart version 1.4.0

**Additional context**
Love the tool !!

devbisme, 2023-01-04:
Thanks for entering this issue!

I created a new `development` branch with a potential fix for this problem. You can install it as follows:
```bash
pip install git+https://github.com/devbisme/kipart@development
```
Let me know if it works for you and then I'll release it on PyPi.

devbisme, 2023-01-05:
Good! I'll release the new version on PyPi.

closed 2023-01-05 by devbisme