devbisme/KiPart #79: Please elaborate KiPart documentation to explain how to run as a script

petemate, 2023-12-26:
I am familiar with python as a scripting language, but I do not understand the part about "running KiPart as a script" in the Usage part of the documentation. After installing using "pip install kipart", I don't know how to proceed. Where do i write "kipart -v" to test the software? Do I run it from a python command promt, from a windows cmd promt, etc? Neither works.

Thank you very much.

devbisme, 2024-03-27:
Apologies for the delay in replying. In a command window, you should be able to type `kipart -v` and get the version number. Let me know if that works.

xobs, 2025-03-21:
I'm running into this issue as well -- I'm not sure exactly how to run kipart:

```
[2:51:14 pm] ~> pip install kipart
Requirement already satisfied: kipart in c:\users\sean\appdata\local\packages\pythonsoftwarefoundation.python.3.10_qbz5n2kfra8p0\localcache\local-packages\python310\site-packages (1.4.2)
Requirement already satisfied: affine>=1.2.0 in c:\users\sean\appdata\local\packages\pythonsoftwarefoundation.python.3.10_qbz5n2kfra8p0\localcache\local-packages\python310\site-packages (from kipart) (2.4.0)
Requirement already satisfied: future in c:\users\sean\appdata\local\packages\pythonsoftwarefoundation.python.3.10_qbz5n2kfra8p0\localcache\local-packages\python310\site-packages (from kipart) (1.0.0)
Requirement already satisfied: pyparsing in c:\users\sean\appdata\local\packages\pythonsoftwarefoundation.python.3.10_qbz5n2kfra8p0\localcache\local-packages\python310\site-packages (from kipart) (3.2.1)
Requirement already satisfied: openpyxl in c:\users\sean\appdata\local\packages\pythonsoftwarefoundation.python.3.10_qbz5n2kfra8p0\localcache\local-packages\python310\site-packages (from kipart) (3.1.5)
Requirement already satisfied: sexpdata==1.0.0 in c:\users\sean\appdata\local\packages\pythonsoftwarefoundation.python.3.10_qbz5n2kfra8p0\localcache\local-packages\python310\site-packages (from kipart) (1.0.0)
Requirement already satisfied: et-xmlfile in c:\users\sean\appdata\local\packages\pythonsoftwarefoundation.python.3.10_qbz5n2kfra8p0\localcache\local-packages\python310\site-packages (from openpyxl->kipart) (2.0.0)

[notice] A new release of pip available: 22.2.2 -> 25.0.1
[notice] To update, run: C:\Users\Sean\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\python.exe -m pip install --upgrade pip
[2:51:18 pm] ~> kipart -v
kipart: The term 'kipart' is not recognized as a name of a cmdlet, function, script file, or executable program.
Check the spelling of the name, or if a path was included, verify that the path is correct and try again.
[2:51:39 pm] ~> python3 -mkipart -v
C:\Users\Sean\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\python.exe: No module named kipart.__main__; 'kipart' is a package and cannot be directly executed
[2:51:43 pm] ~>
```

It was installed using `pip`, but `kipart` isn't a valid command, and `python3 -mkipart` reports that it's not a valid program.

devbisme, 2025-03-22:
I installed and executed `kipart -v` on Ubuntu 22.04 without any problems. It looks like you're using Windows. If so:

>After installing, if the command doesn’t work, ensure your Python scripts directory (e.g., ~/.local/bin on Unix or %USERPROFILE%\AppData\Local\Programs\Python\PythonXX\Scripts on Windows) is in your system’s PATH.

xobs, 2025-03-23:
The directory `%USERPROFILE%\AppData\Local\Programs\Python` doesn't exist.

Investingating using `Get-Command python3.exe` reveals that the scripts directory is actually `%USERPROFILE%\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts` which seems like it probably shouldn't be added to the path, especially since it contains a package hash in there.

I do see lots of files under the `Scripts` directory, none of which I can access from my path but most of which seem to be accessable with `python3 -m[program]`

devbisme, 2025-03-23:
I don't have much insight into this problem, but maybe there's some clue in this Grok output that will lead you to a solution:
https://x.com/i/grok/share/pOJh2FodDQtHdyzrwwJv4GvkG

xobs, 2025-03-23:
I don't understand what that user is saying. They seem to say that Python is installed to the root of the C:\ drive, but if you get the official Microsoft version from their app store (which you get to by typing `python` on a fresh install of Windows) it installs it to somewhere in the user's home directory.

For whatever reason, it adds the Python program and associated executables to `    C:\Users\Sean\AppData\Local\Microsoft\WindowsApps` but doesn't add any installed scripts. I suspect this may be due to sandboxing restrictions, but I haven't been able to find out exactly why.

I can't find any document explicitly stating it, and I'm not very familiar with the Python ecosystem, but it seems as though globally-installed binaries are generally deprecated. They want you to install things with venvs, or to install modules globally and use `python3 -m[module]` to access them rather than installing actual executables. At least, that appears to be how it works for most other modules I'm finding.

devbisme, 2025-03-24:
I thought that using `setuptools` to create the `kipart` Pypi package might be using some old protocol that wasn't allowed on Windows, but there doesn't seem to be any reports of that.

Maybe you could use a different installation tool like `uv` and see if it works.